jcchurch13/Mechaduino-Hardware #1: License and Smoothieboar

arthurwolf, 2016-06-02:
Hey there.

Great project.

Both the hardware and firmware repos are missing a license file as far as I can see.
Once that's fixed and you confirmed it really is Open-Source, if you want we can send you a free Smoothieboard to use/test with your system.

Cheers.

jcchurch13, 2016-06-02:
Hey! Thanks for catching that.  I just added the license files (CC-SA).
Thanks for the offer!  I would love to test the Mechaduinos with a smoothieboard!  Once I get the Mechaduino into production, I'd be happy to send you a Mechaduino for evaluation as well.  
Could you please email me at tropicallabsdc@gmail.com, and I'll send you a shipping address?
Thanks!
Joe

SloMusti, 2016-06-02:
You might find helpful the way I do licensing for open projects in one of repositories here: https://github.com/IRNAS/ToslinkCNC#license

It is rather key to cover all the aspects of the project and I use that standard across repositories, checked by awesome open source layers I spoke to.

Also, I would be keen to test some Mechaduinos also with ToslinkCNC system as shown on above link.

closed 2016-08-22 by jcchurch13