jcchurch13/Mechaduino-Hardware #3: Vector control

markfinn, 2016-07-15:
From what I can understand of the firmware, and from looking at the schematic, you are missing out on field / vector control of the motor.  With vector control you control the torquing current and the "magnetizing current" (think of the current that is 90 degrees out of phase with the torquing current).  This allows for a few things:

- Easy "current advance" and it's more general cousin "field weakening" which is advance past the point needed for just overcoming inductance.  It allows a much higher top speed when torque is allowed to decline.
- Better position accuracy (and lower position latency) by combining the encoder with the driving currents and a motor model through a Kalman filter.  This especially would help here with the 50 pole motor.
- Some amount of force feedback
- Quieter and cooler operation through sine-drive commutation over square wave. I can't tell exactly what you have implemented now from a quick look at the firmware, so this one might not count.

The reason I bring this up in the hardware project is that you are missing motor feedback that is needed to allow this.  The "normal" way to do vector control is to control the phase drive voltages (via PWM) and sense the phase currents.  To do that you would need to add an amplifier to bias and scale the current sense resistors, and ADCs (maybe in the MCU already? I didn't look) that can sample at least around 40 per pole, or 2000 times per rotation on a 1.8 degree stepper.

Another option that may be possible (I'm making this one up as I go along, I don't know if it's been done before or disproven)... Since you are using a bridge driver that self-regulates current, I think it might be possible to implement vector control using a "control the current, read the phase voltage via PWM transitions" model.  This would depend on being able to change the current Vref values at at least a few times per pole to control the phase current directly, and having some digital feedback from the motor phase outputs to know when the A4954 switches between internal PWM modes.  This method would only need a diode or two and some resistors for each of the 4 motor leads.  The big sticking point would be whether the A4954 Vref input has that kind of frequency response.  I don't see anything to give a hint in the datasheet, so it might have to be determined by experiment.

Either of these methods would allow estimating the back-emf of the motor, which gives an estimate of motor position, and when combined with knowing the phase voltage or current give force feedback and the ability to directly control the 2 current vectors occurring in the motor.  

Looking forward to getting some hardware...  I'm glad someone is finally doing this!

jcchurch13, 2016-07-15:
Hi Mark,
We actually are using a form of vector control.  We sinousoidally vary the vref inputs to the a4954 driver as you describe.  Frequency response does not seem to be an issue based on performance.  
-Joe

markfinn, 2016-07-15:
I went through the code enough to understand it now. I should have been more specific about using "vector" vs "field oriented" above.  It's been a while since I've looked at this stuff and I got a little sloppy.

What you're doing is vector based, but you're not really getting field control yet because even though you try to command Iq (torque current) via output() and implicitly command Id to zero, the external controller (The PWM current control loop in the A4954) can't magically make those currents happen in the face of inductance and motor irregularity.  The real Iq ends up lagged and not matching the commanded one, and the real Id ends up nonzero.

I see you've tried "phase advance."  Was that by using a motor model (backemf, resistance, inductance, position) to try to estimate the the actual phase currents and Iq, Id?   If that works well enough, maybe my request for real feedback isn't needed.  I just haven't seen any examples where current sensing was modeled instead of measured before so I assumed it didn't work well. 

Sorry if you've thought about all this already, I'm a little excited about the project.  For the record, my looking into this part came from seeing the FAQ about motor speed only being easy up to 300rpm on the nema17, combined with the link you posted to the 1.8A nema 23 stepper.  It has an inductance of 17mH, so I started getting a bit worried about top speed even at 40 volts.

jcchurch13, 2016-07-16:
Hi Mark,

You are correct:  We are using vector control, but not field oriented control.  There are a lot of advantages that come with field oriented control, but we've found that it would drive the hardware cost and complexity up quite a lot.  By off-loading the current loop to the A4954, we've found that we can deliver pretty impressive performance at an accessible hardware cost.
Thanks for your input! 
-Joe

markfinn, 2016-07-16:
I'm not saying don't use the A4954, I'm saying sense the current.  You
could maybe do it well enough with just AC coupling the shunt into the ADC
channels on the MCU.  That would only be 2 extra caps and 4 resistors.

closed 2016-08-22 by jcchurch13

dzid26, 2020-07-23:
I am just going to link to a discussion where I am coming to the same realization as @markfinn regarding a lack of current sensing and FOC implications. 
https://groups.google.com/g/smart-stepper/c/LuQkAaiCvPs/m/Nzos9Bh_CAAJ  

I am too wondering if it would work to implement a modified (even simpler) open-loop FOC where currents are not sensed but assumed to be correct based on the commanded currents. 
I am thinking that without reading or estimating the real current the method wouldn't be perfect but perhaps a little improvement could be noticeable. At very least the math for field weakening, etc could be reused instead of guessing offsets between phases to weaken the field. 
Additionally, it seems sensor latency is already a limiting factor for higher speed, so it would need to be accounted for separately via phase-advancement trick.