jcchurch13/Mechaduino-Hardware #8: AS5047D Magnetic Encoder Power Management issue.

bitbangr, 2016-09-01:
I was checking over the [Datasheet for the AS50447D Magnetic Rotary Position Encoder](http%3A%2F%2Fams.com%2Feng%2Fcontent%2Fdownload%2F595083%2F1609657%2F320155&sa=D&sntz=1&usg=AFQjCNHJLUOwhbDV-abh1qcGxdTLYPkByQ) 

The section on power management has two configurations. One for 5V operation and the other for 3.3V operation.

For 3.3V operation it states -> "In 3.3V operation, VDD and VREG must be tied together."

In Figure 10 the "3.3 Operation" diagram shows VDD and VDD3V3 tied together with one 100nF cap to ground.

It would seem to me that C17 1uF should be removed from this circuit and pins 11 and 12 of the AS5047D should be tied together.

This could easily be fixed on the current board by a small bridge between C16 and C17 on the top of the board after removing C17.

PS I also [reported this to the Mechaduino Google Group](https://groups.google.com/forum/#!topic/mechaduino/e2BsRLsDtlw)

jcchurch13, 2016-09-01:
We are aware of this discrepancy but have found that it does not seem to have any noticeable effect on performance.  We will make the change in the next rev.  
Thanks!
-Joe

bitbangr, 2016-09-01:
Cool.  I'll close this then.

closed 2016-09-01 by bitbangr