jcchurch13/Mechaduino-Hardware #10: Feature request : Gadgeteer port

arthurwolf, 2016-11-28:
Hey.

As part of the Smoothieboard v2 hardware design, we are planning a very complete series of "extension boards" : goo.gl/XP5iZc
Those boards will all use the "Gadgeteer" socket. It is small, it's a very nice and well designed standards, a lot of boars already exist for it. We also hope to encourage other Open-Source controller board designers ( running Smoothie or otherwise ) to also use these in their hardware.
If you were to add a Gadgeteer port to your board, that'd make it super easy to interface will all of the Smoothieboard v2 hardware boards ( and possibly other boards in the future ).
What do you think ? Maybe for a v2 of your hardware ...

Cheers.

npgoalkeeper, 2018-07-26:
RIP Smoothieboard

arthurwolf, 2018-07-26:
What ? Why ?

DirtyEngineer, 2018-07-26:
Because development seems to have come to a full stand still?

I think this comment stems from this thread. https://github.com/jcchurch13/Mechaduino-Hardware/issues/14#issuecomment-407940180

arthurwolf, 2018-07-26:
> I think this comment stems from this thread. #14 (comment)

Eh thanks, answered there. This is most definitely wrong.

npgoalkeeper, 2018-07-26:
Yea I thought it was dead as the website said the release would be in 2017, and the files haven't been updated in awhile. I like the the thing you're developing with F360 and smoothieware, though.

arthurwolf, 2018-07-27:
It's late, but that's to be expected for such a huge project :)