jcchurch13/Mechaduino-Hardware #11: Compilation error w/ Arduino 1.8.2 IDE

eudoxos, 2017-04-07:
After opening the firmware in Arduino, verification fails with error from gcc. I tried to google things up, nothing shows up. Maybe some problem between keyboard and chair. I am new to mechaduino so maybe I am missing something obvious. I looks as if types were not recognized, but there is no error about #include not being found. These are the first few messages. Thanks for help!

```
sketch/analogFastWrite.c: In function 'syncADC':
analogFastWrite.c:17: error: invalid type argument of '->' (have 'uint16_t')
   while (ADC->STATUS.bit.SYNCBUSY == 1)
             ^
sketch/analogFastWrite.c: In function 'syncDAC':
analogFastWrite.c:24: error: 'DAC' undeclared (first use in this function)
   while (DAC->STATUS.bit.SYNCBUSY == 1)
          ^
sketch/analogFastWrite.c:24:10: note: each undeclared identifier is reported only once for each function it appears in
sketch/analogFastWrite.c: At top level:
analogFastWrite.c:29: error: unknown type name 'Tc'
 static __inline__ void syncTC_8(Tc* TCx) __attribute__((always_inline, unused));
                                 ^
analogFastWrite.c:30: error: unknown type name 'Tc'
 static void syncTC_8(Tc* TCx) {
```

mikbeng, 2017-10-11:
I have the same problem here, did you find any solution to this?

jcchurch13, 2017-10-11:
I would double check that you have the correct board selected in the Arduino IDE (Arduino/Genuino Zero, (Native USB port)).  I've seen similar issues when I've forgotten to select the right board.

-Joe