jcchurch13/Mechaduino-Hardware #14: Feature request: Trinamic drivers

udidnazime, 2018-04-22:
Any word of advice on interfacing trinamic drivers on board instead of the a4954 stepper drivers? This so stepper operation can be quieter.

julidau, 2018-07-23:
could you elaborate further on that thought? Are there any trinamic products specifically that you could suggest ?

npgoalkeeper, 2018-07-24:
I really like the look of the TMC4362A-LA, and they also happen to be cheaper at Digi-key than the TMC4330A-LA, and its information can be found here: https://www.trinamic.com/products/integrated-circuits/details/tmc4361a-la/ as well as the pinout.

julidau, 2018-07-24:
These are nice motion controllers.

They can talk to the AMS5047 directly and implement closed loop stepper control using the encoder feedback. 

**But** they do not contain a driver stage or a stepper driver (current controller). So while they are a nice replacement for the M0+ core (the Atmel SAMD21) on the Mechaduino, they cannot replace the A4954. 

I feel that i should also mention the [PanDrive stepper controllers from Trinamic](https://www.digikey.de/product-detail/de/trinamic-motion-control-gmbh/PD42-2-1140-TMCL/1460-1100-ND/5013387) that are using a similar chip to implement something close to the Mechaduino. While the linked one is quite pricey i think they are a better option then changing the mechaduino hardware.

npgoalkeeper, 2018-07-24:
Would it be possible to use that with the tmc 262 as the stepper driver? or would that drive the price of this device to be over a hundred dollars (im trying to add this to my prusa i3 mks MMU kit. I’ve looked at Trinamic’s PANdrive systems and am not interested to to the insane price).

npgoalkeeper, 2018-07-24:
In that case, what about using a TMC5160-TA? I wasn’t able to find it beforehand on the other comment, however it includes the features of the TMC2660 with an input for an encoder, along with the features of the TMC4362A-LA. along with that, it is chainable via UART on the datasheet. Along with that, you can add extrernal mosfets, making this closed loop stepper sytem seem more attractive to cnc applications. e. g. one large stepper board chained to multiple smaller, cheaper ones. This IC is also cheapper than the TMC262 and TMC4362-LA at digikey.

julidau, 2018-07-25:
well that chip sure looks great. I really wish i would have known about it before making a driver based on the TMC2660 :) 

That chip only has a few drawback though:
- The Encoder interface accepts quadrature encoders only. It makes sense (most stepper motors come with quadrature encoders either build in or added) but is different from the TMC4362-LA, which also accepts absolute position encoder feedback. Not that one would need that anyway, quadrature should be fine for most applications. A replacement for the AMS5047 would be the [AMS5601](https://www.digikey.com/short/j5v2mn) with a quadrature interface.
- The external mosfets are not optional, they are required. The chip only contains mosfet drivers. This is not that big of a deal as low power mosfets can be had for a buck each, but one will need 8 of them.

I am quite confident that, using the TMC5160-TA, a drop-in replacement for the mechaduino can be made. I'm not sure about doing it on two layers, but it might be possible. I unfortunately lack the time to do so :(

npgoalkeeper, 2018-07-25:
Huh. So to get a better, cheaper motion controller/ driver, you have to sacrifice some precision? Not like over 10 thousand positions would affect something that only steps to 400. When will the TMC 2660 variant of this driver be available? Will it still maintain the SAMD21G18A-AUT? No Rush.

julidau, 2018-07-25:
just in case it seemed that way: i am not affiliated with the mechaduino project :wink:.  I just happen to own one... 

As it seems this project died over a year ago, i don't think there will be any version using a trinamic driver. I was working on a different driver style using the TMC2660 (one in a format similar to [this one](https://www.aliexpress.com/item/Dawupine-Schrittmotorsteuerung-Leadshine-DM856-2-phasen-schrittmotor-57-86-Digitale-Schrittmotortreiber-20-80-Vdc-1A-bis/32835988065.html?spm=a2g0x.search0104.3.43.6d5a8fb7TaQb5c&ws_ab_test=searchweb0_0%2Csearchweb201602_2_10320_10152_10151_10065_10321_10344_10068_10342_10547_10343_10322_10340_10548_10341_10696_10084_10083_10618_10304_10307_10820_10821_10302_10843_10869_10868_10059_100031_10319_10103_10624_10623_10622_10621_10620-normal#cfs,searchweb201603_46,ppcSwitch_2&algo_expid=c5cd6080-6b93-4cec-9fca-baf0c47b8db8-6&algo_pvid=c5cd6080-6b93-4cec-9fca-baf0c47b8db8&priceBeautifyAB=0)). 

Regarding the encoder and the loss of precision: The AMS5047 is currently hard to come by (mouser has a few left i think) and the one i proposed was the "next best thing". Whether that would be the optimal choice i cannot say for certain. AMS might have 14bit encoders with a quadrature interface in their lineup. I was not looking super hard to be honest…

Cheers :)

npgoalkeeper, 2018-07-26:
Wow. Now I feel stupid. So um, thanks for your help. It seemed like you were putting an awful lot of effort into your replies, and I thank you for that. It seems that a lot of things 3D-Printing as far as electronics go have died in the past year. The Smoothie-V2 seems to be dead, as well as many other awesome boards. With duet being at the, top they really have no reason to innovate and the revolve board will be way too expensive for a standard user. This makes me curious of someones gonna end up making a REALLY successful kick-starter that includes 3D printing electronics. Anyways, thanks again, and one last thing: Are you gonna make that motor driver design public on easyEDA? I think they guys over at reprap would have a field day over that.

julidau, 2018-07-26:
No worries! I just wanted to lower the expectations a bit :)

Yes, i was planning on making the design publicly available after fully testing it out. And who knows? Maybe i will come around to making a driver board thats screws to the back of the stepper like mechaduino does. I too am curious what the future holds...

npgoalkeeper, 2018-07-26:
I just did some reading and found out that the TMC5130 has the same features as the 5160, however it has internal MOSFETS, so you may want to check that out at a later date.

arthurwolf, 2018-07-26:
@npgoalkeeper  Smoothie v2 is most definitely not dead, both Smoothie v1 and v2 are very actively developed, it's just we are doing extremely ambitious things at the moment so there are longer development time spans than usual.  I've just sent the gerber for the first proto of v2 to the board house, and we already have protos for v2-mini and v2-pro. And v2 firmware can laser cut and cnc mill at this point.

npgoalkeeper, 2018-07-26:
Sooo its just issues with FW and getting the pcb's manufactured at this point?

arthurwolf, 2018-07-27:
Mostly yes.

npgoalkeeper, 2018-07-28:
So I found the AMS AS5311 magnetic hall effect IC, and it uses an ABI interface (looks like it can be programmed to ABN for the TMC5130) and all it requires is an SDI/SCL interface to program. I’m going to try and make something close the mechaduino, and it should in theory only need a location to get to.

julidau, 2018-08-22:
hey,
how is your project coming along ?

npgoalkeeper, 2018-08-23:
Well, I'm going to use the TMC5130 because of its microplyer capabilities. The driver will (normally) be with 16 bit precision with interpolation. I'm probably also going to use it in S/D mode. I will also use the TMC4361a. I could just simply use the TMC5130 by itself, however the TMC4361a has true closed-loop operation, along with a 1.8v on-board regulator. Everything will use 3.3v logic. The encoder should require little coding, and thats pretty much it, other than that I'm unsure if I should use SPI or S/D mode for communication between the the TMC4361a and TMC5130 and that I'll maintain the A/B/N encoder interface. I may not use the original form factor, as I'm planning to use the maximized configuration with any and all inputs/outputs with pull-up or a pull-down resistor. Unfortunately, the system will only be able to handle up to 1.4a with 2.5a peaks, however I think that with good cooling for bigger machines, it'll be fine. After all, with a closed-loop interface, coolstep and dcstep should allow the motors to run at a lower current in normal conditions, and near-none when idle.

julidau, 2018-08-31:
I took a look and it looks great. I actually played with this a little myself but your approch looks a lot better! 

I did a smaller "version" of the mechaduino using the TMC5130A and ordered a prototype, but this one does not include an encoder since it was made for a NEMA11 motor. 
If someone wants to check it out it can be found here: https://aisler.net/p/QVYWBVJL

I will test that one in a month and maybe tackle a NEMA17 version if i got more familiar with the 5130A. Till then i wish you @npgoalkeeper good luck with your version!

npgoalkeeper, 2018-09-03:
After looking at it some more, and coming up with new ideas, and seeing trinamic's new tmc2160 stepstick, i have determined there is no use in using a tmc5130. which is why im switching to the tmc2160. Also im probably going to also have a 2 header layout like the uStepper. Oh and yes, i will keep the as5047d since the tmc4361a does not require a quadrature interface. Might make a version using the  tmc389 ic as well, might be fun to play with.

PBscoots, 2018-09-04:
Just lurking here, but its awesome you guys are reviving this project. Excited to see the progress in this revival and some ultra silent mechaduinos. :) I would help but am worthless at board level.

npgoalkeeper, 2018-09-06:
With the TMC5160, I need to use external mosfets. Any suggestions? They need to be N-channel. For my design, I will copy the TMC2160 silentstepsticks for now (well a variation).

RGD2, 2018-09-06:
> My goto method of searching for mosfets is going on digikey and filtering
> the mosfet category for new products. That turned up with this one:
>
> https://www.digikey.com/product-detail/de/on-semiconductor/NVMFD5C478NLT1G/NVMFD5C478NLT1GOSTR-ND/8636571
>
> gate charge looks ok, Ron should be fine as well. But i dont know what
> max. Voltage you want to go to. Maybe there are smaller packages for lower
> voltages...

Good trick for also is to pick a package appropriate to the power
dissipation you expect and filter on that first.

Remember also that there’s a fair bit of “specmanship” going on: they
usually won’t do both their full current and/or full voltage ratings at any
reasonable switching frequency.

What really matters is the switching energy, as well as their power
dissipation when hard-on. The heating they take as they change state is
usually where most of the heat they cop comes from, assuming Rds(on) is low
enough.

Finally, don’t forget to check how much current drive they need to
charge/discharge the gate capacitance quickly enough. Specs might be good,
but if driven too weakly they’ll overheat during the slower switching time.

Oh, and put a small smd resistor in series with the gate - best value is
the lowest which suppresses any VHF “ringing” when it’s switched: or else
that can drive the thing through switching multiple times and multiply the
heating it takes for each switch transition, which can also kill them.
Inductance is bad in the gate drive circuit, but a little is unavoidable.
The gate is capacitance, so you need enough resistance to dampen out the Q
of the resulting probably MHz RLC series resonator. Expect Low tens of Ohms
or so.

Get all the above right (better yet, benchmark with a fast scope) and it
will run cool and reliable.

It all gets more critical as you go up in power, and for very high power
drive (kW motors) there is even more to consider. (Snubbed on the output,
secondary breakdown from too fast dv/dt on the output, etc). Consult a good
book on power electronics if you’re interested.

Good luck!

npgoalkeeper, 2018-09-07:
So basically choose Transistors with a low voltage drop and power dissipation? I don't have a fast scope, much less a scope 8n general, just a voltmeter. I am not a professional.

RGD2, 2018-09-07:
Pick by package, then the voltage you need, and filter also by how much
drive current the controlling chip has: you use the effective Cgs figure
(and gate thresholds) for this: you want to know how quickly the the
driving chip can change the gate voltage between on and off and vice versa.
When you know how long this is (probably so many ns) you can guess how hot
it’ll get if it’s switching so much drive current each time, and at what
supply voltage. You should assume it cops practically half the voltage
times the current in power, for however many ns it was: this will give an
energy figure: plug in the switching frequency you expect at worst case,
and you’ll know how much power on average, excluding what it cops when it’s
just on steadily: that latter figure is just Rds(on)*I^2, add it too. The
total is how much power it will really have to handle continuously: you
compare that to the thermal resistance the package has, and get the
operating temperature it will be, using what worst-case environment
temperature it will heat up hotter then, and then see if it falls in the
limits of the device.

The switch energy is often nicely given as so much gate charge instead of a
capacitance: you just divide it by the driving chips’ gate drive current to
figure out how long it takes to charge the gate in that case.

All this will let you separate out the parts that are actually crap but
just “look good” : whose manufacturers hope people will just look at the
current ( or rds(on) ) and voltage and pick the cheapest. They’re not cheap
if they keep overheating and dying in no time in use!

You’ll find that the *effective* current each transistor will survive is a
fair bit less than the maximum specified.
So a good way to do this is to download a spreadsheet of all the
*available* parts which has the parameters you need to figure this stuff
out: then write a bit of python or so (use pandas to sort/filter and
calculate new columns) or otherwise whatever spreadsheet type software you
like, and figure out what the operating temperature vs max temp is for each
one: bigger gap means longer life, compare that against cost and you can
pick one thats “good enough” and reasonably priced.

Don’t worry about not being a professional: so many “pro” products don’t
bother with this and they just build it, then bench test with a
thermometer, and reduce current until it stops overheating... which should
still be done anyway.

Other way you could go about this is to pick a footprint that allows
different possible transistors to be installed (be careful, not all have
the same pin assignments!) and then leave it up to the builder: this might
be more appreciated long term anyway, since that way changing transistors
when new ones come out or the current one gets eol’d is easier. The latter
spacing will make the boards a little bigger, but that will also help with
heat dissipation too.

When the transistor is off, it’s not dissipating any power because no
current. When it’s all the way on, it’s “ON” resistance is very low anyway:
that figure makes for great looking high current spec that makes the chip
look good: but it’s the power dissipated during switching, which **depends
on what is driving it** that decides how hot it’ll really get, which really
is what limits the current.

If you just filter for the lowest apparent dissipation, you may end up with
transistors with big gate capacitances (because the bigger gates allow a
lower Rds(on) figure) which actually take longer to switch, and which may
keep overheating because of that, unless only driving much less current
than hoped.

All this is why “integrated drive” chips are so popular.

npgoalkeeper, 2018-09-07:
"Integrated Drive" IC's have too low of an amperage rating for CNC applications. On second thought, I will copy the TMC5160 reference design. Updates on the project will be posted here: https://github.com/npgoalkeeper/Closed-loop-stepper/blob/master/README.md
(I really need a name for it)

npgoalkeeper, 2018-09-14:
Just a bit of an update, I've been talking with trinamic and they have convinced me to simply use a tmc2130, with it the tmc2160 coming out soon.

npgoalkeeper, 2018-09-15:
[image]
Its begun.

npgoalkeeper, 2018-09-17:
So I did a small amount of research on the VHF ringingyou spoke of, and found a thing called a snubber network. Are therea any downsides to using an RCD snubber? It seems to do the same exact thing as the resistor, however it acts as more "active" filter that completely removes the inductance that causes this phenomenon. However, it seems like more than anything my main concern should be layout as far as efficiency goes when I make the tmc2160 version.

RGD2, 2018-09-18:
Yes, Snubber with diode is better - just more complex, more expensive, and
needs to be the right one.

 But yes, even just one resistor is acting as a snubber, just a simple
linear one.

Diode is non-linear, but so are ceramic caps not of np0 or c0g types — x7r
or any code other than the first two have capacitance which changes (quite
a lot!) with voltage — bad if you want a filter to stay tuned, but the
nonlinearity is actually good for avoiding resonance —   It is impossible
without that linearity. The whole point of a snubber is to prevent
unintended resonance and Q factor from causing a destructive over-voltage
condition.

The simple gate resistor is just enough to drop the Q of the RLC resonator
that exists because there is too little R in the gate pin, and the gate pin
+ pcb trace has some small inductance.

Then you can get into active circuits with transistors set to help turn off
the transistor faster by providing a lower impedance route to ground right
next to the main mosfet gate. All this becomes increasingly important as
you get up to the higher power levels / voltages and currents, since the
energy  in the ringing is a proportion of the gate drive power, and bigger
transistors mean much large gate capacitance, and require higher gate drive
current, and more careful handling of the snubbing.

amoose136, 2018-09-22:
Just a note on the TMC2130-LA: my printer uses an Einsy Rambo 1.1a board as the driver and even though the RMS current is rated for 1.2A, in practice if you print with Marlin setting current at 800mA, the motors will reach a steady state temp around 50C. More current does not seem like a good thing here but I could see that maybe a longer Nema 17 could handle more and use more in milling applications with no issues.

npgoalkeeper, 2018-09-22:
Well I mean, this is mainly just for experinentation. Might be fun to use nema 17's in a milling machine just for wood or MDF or whatever.

npgoalkeeper, 2018-12-09:
So here's me a couple weeks later, after some testing. A friend and I have done some research and written the software, so we are going to make a motion controller that consists of A STM32F401 MCU paired with 2x TMC4361a's and 2x TMC2160's paired with 12a MOSFETs. As of right now, I do not want to share the link to access the project on EasyEDA

Psylenceo, 2019-04-19:
I keep screwing this up. Here is 2 boards I made up to allow more user functionality, use a TMC5130 or TMC2130, and eitehr a atmega32u4 or a atmega328PB. With a 3rd board in progress to offload the driver ic and instead have the user use a break out borad like the TMC2130-BOB, TMC5130-BOB, or TMC5160-BOB.

32u4 board - https://easyeda.com/kenm/Intele-step
328PB board - https://easyeda.com/kenm/mechaduino-328pb